rcore-os/rCore-Tutorial-Book-v3 #213: 请问如何在 Mac m 系列芯片的笔记本上配置环境呢?

Ucag, 2024-06-16:
我想在 macbook pro m2 上部署开发环境, 不过我看了很多环境准备文档都是针对 x86_64 的, 请问有可能在 m 系列上部署一套开发环境吗?

Angus-Liu, 2024-10-07:
Mac（m 系列芯片） 教程整体参考：http://rcore-os.cn/rCore-Tutorial-Book-v3/chapter0/5setup-devel-env.html 其中改动地方如下：

1. 安装 Rust：
    - 访问官网获取最新的安装方式：https://www.rust-lang.org/zh-CN/tools/install
    - 切换到 nightly 版本
        
        ```bash
        rustup install nightly
        rustup default nightly
        ```
        
    - 安装相关软件包
        
        ```bash
        rustup target add riscv64gc-unknown-none-elf
        cargo install cargo-binutils --version=0.3.3
        rustup component add llvm-tools-preview
        rustup component add rust-src
        ```
        
2. 安装 qemu：
    - 安装（不需要教程中的设置）：brew install qemu
    - 查看版本：qemu-system-riscv64 —version （没有 qemu-riscv64 不影响，Mac 不支持用户模式模拟）
3. 安装终端服用工具 Tmux：`brew install tmux`
4. 运行项目：
    
    ```bash
    git clone https://github.com/rcore-os/rCore-Tutorial-v3.git
    cd rCore-Tutorial-v3/os
    make run
    ```

Ucag, 2025-06-20:
非常感谢!!!!!

closed 2025-06-20 by Ucag