rcore-os/rCore-Tutorial-Book-v3 #242: 在编译指导书生成html版本时遇到的问题

zmhab, 2025-04-28:
如题，我查阅了相关的文档和博客，配置了编译实验指导书所需要的环境，但是在最终的编译过程中遇到了如下所示的Error：
```shell
$ make html

Running Sphinx v8.2.3
loading translations [zh_CN]... done

Theme error!

Versions
========

* Platform:         linux; (Linux-6.8.0-1020-raspi-aarch64-with-glibc2.39)
* Python version:   3.12.3 (CPython)
* Sphinx version:   8.2.3
* Docutils version: 0.21.2
* Jinja2 version:   3.1.6
* Pygments version: 2.19.1

Last Messages
=============

None.

Loaded Extensions
=================

None.

Traceback
=========

      File "/home/zhang/Tools/Python/lib/python3.12/site-packages/sphinx/theming.py", line 228, in create
        raise ThemeError(__('no theme named %r found (missing theme.toml?)') % name)
    sphinx.errors.ThemeError: no theme named 'furo' found (missing theme.toml?)

The full traceback has been saved in:
/tmp/sphinx-err-pwnwqn04.log

To report this error to the developers, please open an issue at <https://github.com/sphinx-doc/sphinx/issues/>. Thanks!
Please also report this if it was a user error, so that a better error message can be provided next time.
make: *** [Makefile:17: html] Error 2
``` 
这是bug吗？如果是的话，应该如何修复呢？

另外，我尝试使用`make pdf`命令编译实验指导书的PDF版本，结果如下所示：
```shell
$ make pdf

Running Sphinx v8.2.3
loading translations [zh_CN]... done

Sphinx error!

Versions
========

* Platform:         linux; (Linux-6.8.0-1020-raspi-aarch64-with-glibc2.39)
* Python version:   3.12.3 (CPython)
* Sphinx version:   8.2.3
* Docutils version: 0.21.2
* Jinja2 version:   3.1.6
* Pygments version: 2.19.1

Last Messages
=============

None.

Loaded Extensions
=================

None.

Traceback
=========

      File "/home/zhang/Tools/Python/lib/python3.12/site-packages/sphinx/registry.py", line 185, in preload_builder
        raise SphinxError(
    sphinx.errors.SphinxError: Builder name pdf not registered or available through entry point

The full traceback has been saved in:
/tmp/sphinx-err-xm2_67pz.log

To report this error to the developers, please open an issue at <https://github.com/sphinx-doc/sphinx/issues/>. Thanks!
Please also report this if it was a user error, so that a better error message can be provided next time.
make: *** [Makefile:25: pdf] Error 2
``` 
这个问题又该如何解决呢？

wyfcyx, 2025-04-28:
我本地使用的sphinx版本是4.3.2，请先切换成这个版本看能否解决问题。

zmhab, 2025-04-29:
> 我本地使用的sphinx版本是4.3.2，请先切换成这个版本看能否解决问题。

切换至4.3.2版本后不符合最低版本要求（最低版本要求5.0）：
```shell
$ make html

Running Sphinx v4.3.2
loading translations [zh_CN]... done

Sphinx version error:
The sphinxcontrib.applehelp extension used by this project needs at least Sphinx v5.0; it therefore cannot be built with this version.
make: *** [html] Error 2
``` 
然后切换至5.0版本，出现如下错误：
```shell
$ make html

Running Sphinx v5.0.0
loading translations [zh_CN]... done
making output directory... done

Theme error:
no theme named 'furo' found (missing theme.conf?)
make: *** [html] Error 2
``` 
是代码本身的bug吗？

zmhab, 2025-05-01:
> 可能你的平台有一些独特之处吧，我的环境并没有5.0的最低版本要求，4.3.2还是能正常编译的。 关于一开始sphinx=8.2版本找不到furo主题的问题，可以在`source/conf.py`文件中注释掉`html_theme = 'furo'`这一行使用默认主题再试试。

OK, let me have a try. Thank you!

closed 2025-05-01 by zmhab